Field-Robotics-Japan/UnitySensors #16: Map GameObject is missing in Sensors Scene

JaVo93, 2022-01-11:
Hi,

in the Sensors Scene the map GameObject is missing as I saw in the Gif. This leads to the error when users want to visualize the LaserScan in rviz: For frame [scan_link]: Fixed Frame [map] does not exist.

Further the recursive clone of the other two projects does not work.

Best regards

RyodoTanaka, 2022-01-13:
Thank you for the question.
I missed to write the launch file to publish TF info from `map` to `velodyne` and `scan_link`.
The solution is, add following code into your launch file.
```xls
<launch>
  <node pkg="tf" type="static_transform_publisher" name="velodyne_tf" args="-2 0 0.6378 0 0 0 map velodyne 100" />
  <node pkg="tf" type="static_transform_publisher" name="scan_link_tf" args="-1.907 0 0.32 0 0 0 map scan_link 100" />
</launch>
```

Bests,

RyodoTanaka, 2022-05-18:
@JaVo93 
Did you solve the problem ?
If so, please close this issue or comment your updated problem.

If you do not update this issue until this month (2022/05), I will close this as solved issue.

closed 2022-06-01 by RyodoTanaka