Field-Robotics-Japan/UnitySensors #17: Now able to visualize pointcloud data

Hrithik-verma, 2022-01-17:
[ WARN] [1642414105.183307373]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.velodyne_pointcloud.message_filter] rosconsole logger to DEBUG for more information.
[ WARN] [1642414105.183512721]: MessageFilter [target=map ]:   The majority of dropped messages were due to messages growing older than the TF cache time.  The last message's timestamp was: 686.699280, and the last frame_id was: velodyne

This is warning I'm getting while launch velodyne_pointcloud. And in rviz pointcloud2 data is empty.

RyodoTanaka, 2022-05-18:
This error occurs from ROS side problem.
The problem comes from following reason.
1. You do not set `use_sim_time` parameter as `true`.
   To do that, you just run `rosparam set use_sim_time true` after the roscore started.
2. You mistakenly set target frame on rviz to `map`.
  To fix that, you need to set `velodyne` to the target frame.

robograffitti, 2022-06-28:
I created #20 related to this issue.

closed 2023-10-26 by RyodoTanaka