Field-Robotics-Japan/UnitySensors #18: 64 Channel Lidar

sawan-ES, 2022-01-17:
Hello,
I figured that [RotateLidar](https://github.com/Field-Robotics-Japan/UnitySensors/blob/6ce696958fbc0f761c48421c2f21c51b417f0319/Scripts/Runtime/Lidar/RotateLidar.cs) script allows us to change the number of layers. I want to know that how can we use this to make our lidar 64 channel instead of 16 channel? How should i modify the [VLP16Publisher](https://github.com/Field-Robotics-Japan/UnitySensorsROS/blob/07b53092fae37c514c6485a2e60375cd419d7e68/Scripts/Runtime/Velodyne/VLP16Publisher.cs) script to do so? Do i also have to modify the [VLP16Serializer](https://github.com/Field-Robotics-Japan/UnitySensors/blob/6ce696958fbc0f761c48421c2f21c51b417f0319/Scripts/Runtime/Velodyne/VLP16/VLP16Serializer.cs) script?

RyodoTanaka, 2022-05-18:
@sawan-ES 
To utilize 64 channel instead of 16 channel, you need to do following.
1. Change number of layers parameter on the RotateLidar script.
2. Create 64 channel serializer script OR just publish the topic as any type you want.
   To create 64 channel serializer script, you need to check Velodyne 64 channel serialization protocols (check the following link)
   https://velodynesupport.zendesk.com/hc/en-us/articles/115001950374-HDL-64S3-Data-Packet-Format

sawan-ES, 2022-05-23:
what is meant by "just publish the topic as any type you want." 
Will this method not require serialization? I want to publish it as [sensor_msgs/PointCloud2](http://docs.ros.org/lunar/api/sensor_msgs/html/msg/PointCloud2.html)

RyodoTanaka, 2022-06-01:
The serialization is needed.
But the serialization is not for sensor_msgs/PointCloud2 but velodyne_msgs/VelodynePacket.

The VLP16 example is publishing velodyne_msgs/VelodynePacket, then on the ROS node (velodyne_pointcloud) is converting  it into sensor_msgs/PointCloud2.

closed 2023-10-26 by RyodoTanaka