Field-Robotics-Japan/UnitySensors #19: Velodyne wrong orientation.

argolomb, 2022-03-10:
Using the Velodyne LiDAR the point cloud was incorrect about orientation, maybe it's because of the different coordinate system between ROS and Unity, I try to make some changes on Rotate Lidar Scrips but without success.  A screenshot of the same view on unity was expected on Rviz but the objects appear to be mirrored or flipped. Ex. The red ball should be on the right on RVIZ but it's on left. 
[image]
.

babindemers, 2022-05-12:
Change Rotate Dir to CW in the Rotate Lidar Script in the Unity Inspector window. It seems to have fixed the problem for me.

RyodoTanaka, 2022-05-18:
@argolomb 
The solution is completely same as @babindemers wrote.

@babindemers 
Thank you for your comment !!

RyodoTanaka, 2022-06-01:
@argolomb 
Have you notice the reply ?
I think the solution is already reviewed.
Thus, I will close this issue after this month (June 2022).

If you have more problems, please re-open it.

Best,

argolomb, 2022-06-02:
Sorry for my late. 
@babindemers thank you.

closed 2022-06-15 by argolomb