Field-Robotics-Japan/UnitySensors #29: FRJが見つからないというエラーが出ます。

moriitkys, 2023-06-18:
Assets\UnitySensorsROSAssets\Scripts\Runtime\RGBCamera\RGBCameraPublisher.cs(23,11): error CS0246: The type or namespace name 'FRJ' could not be found (are you missing a using directive or an assembly reference?)

Autumn60, 2023-06-21:
現在リポジトリを改修中で、READMEに載っている手順に従っていても、正しいディレクトリ構成になっていない可能性があります。

旧UnitySensorsがUnitySensorsArchiveに
旧UnitySensorsROSがUnitySensorsROSArchiveに
旧UnitySensorsROSAssetsがUnitySensorsに
それぞれリポジトリ名が変更されているため、旧UnitySensorsが正しくインストールされていないのだと思います。

本来のディレクトリ構成は以下のようになります(重要でないところは省略します)。

- [本リポジトリのRoot](https://github.com/Field-Robotics-Japan/UnitySensors/tree/master)
    - Assets
        - UnitySensors（旧[UnitySensors](https://github.com/Field-Robotics-Japan/UnitySensorsArchive)(現UnitySensorsArchive)のインストール先）
        - UnitySensorsROS（旧[UnitySensorsROS](https://github.com/Field-Robotics-Japan/UnitySensorsROSArchive)(現UnitySensorsROSArchive)のインストール先）
        - UnitySensorsROSAssets

なので、ひとまず動作確認のみが目的であれば、本リポジトリを--recursiveオプション無しでクローンし、Assets/UnitySensorsフォルダとAssets/UnitySensorsROSフォルダを、それぞれ[UnitySensorsArchive](https://github.com/Field-Robotics-Japan/UnitySensorsArchive)と[UnitySensorsROS](https://github.com/Field-Robotics-Japan/UnitySensorsROSArchive)リポジトリのmainブランチで上書きすることで動作が確認できると考えられます。

Autumn60, 2023-06-21:
[tempMasterブランチ](https://github.com/Field-Robotics-Japan/UnitySensors/tree/hotfix/tempMaster)にて、上記手順を実行済みの、スタンドアロンな状態のプロジェクトを用意したのでそちらを試していただけると幸いです。

moriitkys, 2023-06-25:
お返事遅くなり大変申し訳ないです。動作確認できました。
お忙しいところ、ご対応いただき大変ありがとうございました。

closed 2023-06-25 by moriitkys