Field-Robotics-Japan/UnitySensors #32: Error in ROS2

lidiaxp, 2023-07-16:
Hey, I am using your code in my research. The last update in the code add some features but created some errors to ROS2 simulations. 

[image]

Autumn60, 2023-07-25:
Could you try this "hotfix ros2 header" branch?
I think it has been fixed.

lidiaxp, 2023-08-25:
Actually I am using the package adding by the git url, is it possible to set the branch in this way?

Autumn60, 2023-08-25:
Yes, you can download packages from a specific branch by appending "#{BRANCH_NAME}" to the URL.

Example:
・UnitySensors pkg from develop branch

https://github.com/Field-Robotics-Japan/UnitySensors.git?path=/Assets/UnitySensors#develop

・UnitySensorsROS pkg from develop branch

https://github.com/Field-Robotics-Japan/UnitySensors.git?path=/Assets/UnitySensorsROS#develop

If you've already downloaded this pkg from another branch, you don't need to remove old version.

closed 2023-10-05 by Autumn60