Field-Robotics-Japan/UnitySensors #33: The UNKNOWN_OBJECT_TYPE TextureToPointsJob.pixels has been deallocated. All containers must be valid when scheduling a job.

hushuaibo, 2023-07-20:
The UNKNOWN_OBJECT_TYPE TextureToPointsJob.pixels has been deallocated. All containers must be valid when scheduling a job.

hushuaibo, 2023-07-24:
When I wanted to use UnitySensors, this issue occurred “The UNKNOWN_OBJECT_TYPE TextureToPointsJob.pixels has been deallocated. All containers must be valid when scheduling a job.” and LivoxMid70 was unable to hit points on my own obstacles

Autumn60, 2023-07-26:
Several bugs have been fixed in https://github.com/Autumn60/UnitySensors/pull/1
Could you try "develop" branch?

RyodoTanaka, 2023-11-08:
We need to represent @hushuaibo bug situation.
If anyone meets same error, please report it.

Autumn60, 2023-12-08:
I have found a way to fix it, but I am currently working on other tasks and cannot submit a PR. So I will just describe how to fix it.

After `AsyncGPUReadback.Request()`
Call  `AsyncGPUReadback.WaitAllRequests()`

closed 2024-01-12 by Autumn60