Field-Robotics-Japan/UnitySensors #55: IMU acceleration is affected by scale

sgrsn, 2023-08-09:
In the IMU calculation, we are using Transform.InverseTransformVector to add the gravity, but this function is scale sensitive, so if the object's scale has been changed, the acceleration values will not be calculated correctly.

If the scale is set to 1, there is no problem.
[image]

If the scale is changed to 0.1, the gravity of the corresponding axis is calculated as 10 times.
[image]

Autumn60, 2023-08-09:
Fixed this issue in #57.

sgrsn, 2023-08-30:
@Autumn60 
I have confirmed another problem. InverseTransformVector has the effect of shifting the angle if the scale of each axis is different. For example, when tilted by 45°, the acceleration of the two axes does not have the same magnitude.
I think we should consider a method that does not use Transform.InverseTransformVector.
[image]
[image]

RyodoTanaka, 2023-11-08:
This issue looks solved.
I close this.

PR #57 (2023-08-09): Fix/imu acc calc
Fix #55
PR #63 (2023-10-26): Fix/imu acc calc
Fix #55
closed 2023-11-08 by RyodoTanaka