Field-Robotics-Japan/UnitySensors #64: Depth camera values are wrong after build.

DaiGuard, 2023-10-02:
# Overview

When I play in Unity, I can subscribe to the point cloud data as I expect, but I am not getting the correct point cloud that I can subscribe to from the application after I build it.

# Environment

Ubuntu 20.04
Unity 2022.3.5.f1
ROS TCP Connector: 0.7.0-preview
UnitySensors: 0.1.1
UnitySensorsROS: 0.1.1
URDF Importer: 0.5.2-preview

# Capture Image

* Play Mode
[image]

* After Build

[image]

Autumn60, 2023-10-05:
Thanks for the issue report.
We were not able to confirm the operation of the built application.

It is similar to when the depth value is inverted.

Could you try changing [line 48 of Color2Depth.shader](https://github.com/Field-Robotics-Japan/UnitySensors/blob/master/Assets/UnitySensors/Runtime/Shaders/Color2Depth.shader#L48) as follows.
~~~
float distance = length(viewPos) / _F;
~~~

DaiGuard, 2023-10-12:
@Autumn60 

Thank you for your response.
I was trying i out right away, but the depth data is still wrong in the build application.

## Change code
[image]

## Capture image

- Play Mode
[image]

- After build
[image]

panagelak, 2023-10-27:
@DaiGuard  @Autumn60 hello, i fix it by copying this shader 

https://github.com/Field-Robotics-Japan/UnitySensors/tree/master/Assets/UnitySensors/Runtime/Shaders 

into my project assets,

the go to Edit -> Project Settings -> Graphics and include this shader to the always included shaders.

i suppose this shader is not linked in the script but rather using a shader.find method

cheers

RyodoTanaka, 2023-11-08:
This solution must be added to trouble shooting on README.
I will create PR.

RyodoTanaka, 2023-11-08:
I added this discussion into README trouble shooting.
Thank you @panagelak and @Autumn60 !!

closed 2023-11-08 by RyodoTanaka