Field-Robotics-Japan/UnitySensors #65: Mac os render error without psize

LiangHuangBC, 2023-10-04:
Running on Macos, with tons of warning:
Drawing with MeshTopology.Points, yet the vertex program 'Sprites/Default' does not have PSIZE output.

And page is messy with wrong points
[image]

Autumn60, 2023-10-05:
Thanks for the Issue report.

It seems to same issue as https://github.com/keijiro/Pcx/issues/71.

There are two possible solutions at this time

(1) Set [Material] to [None] in VelodynePointsMeshGenerator.cs.

[image]

(2) Deactivate the VelodynePointsMesh object and change the [Visualize Mode] of VelodyneVisualizer.cs, which is attached to VelodyneSensor, to [ALWAYS].

[image]

RyodoTanaka, 2023-11-08:
This issue looks solved.
I will close it.
If you need more discussion, please do not hesitate to re-open it.

closed 2023-11-08 by RyodoTanaka