Field-Robotics-Japan/UnitySensors #74: IMU Noise model

RyodoTanaka, 2023-12-04:
We do not set any noise model for IMU.
The usual IMU they have several noise model.
I think following noise model is needed mainly.

1. Gussian Noise
2. Draft Noise
  This noise could be represented by using time scale.
3. Temperature Noise

If anyone have any other idea, it is very welcome to write them on here.

qoopen0815, 2024-01-20:
I'm interested in implementing the proposed noise model, but I'm a little unsure where to stick it in my code. Could you please throw a hint as to which part of the codebase is needed?

Autumn60, 2024-01-20:
@qoopen0815 

Thank you for your interest.
Currently, codes for the noise model are located in the following directory

https://github.com/Field-Robotics-Japan/UnitySensors/tree/master/Assets/UnitySensors/Runtime/Scripts/Utils/Noise

Also, the namespace would be [UnitySensors.Utils.Noise](https://field-robotics-japan.github.io/UnitySensors/namespaceUnitySensors_1_1Utils_1_1Noise.html).

qoopen0815, 2024-01-20:
@Autumn60 
Thanks, I found the code!

Quick question:
Any reason the noise model, which seems ready to roll, hasn't been integrated into the IMU yet? Is there a specific reason or is the implementation approach still up in the air?

Autumn60, 2024-01-20:
That's because the implemention approach was not set in stone.

I would like to have a framework that allows us to combine multiple noise models in the future.
(e.g., Gaussian noise + drift noise, Gaussian noise + temperature noise, etc.)
Given this, I believe it is necessary to have a common interface for all noise model classes, like [PointCloud](https://github.com/Field-Robotics-Japan/UnitySensors/tree/master/Assets/UnitySensors/Runtime/Scripts/Data/PointCloud) data.

Autumn60, 2024-02-09:
https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model

RyodoTanaka, 2024-02-19:
@Autumn60 
The ethz-asl's link you've found looks very interesting !
It seems we also could be possible to implement the noise model 😎