Field-Robotics-Japan/UnitySensors #114: Hello, after adding lights, the sensors cannot properly scan objects after packaging , What went wrong and do you have any good suggestions。

Autumn60, 2024-02-07:
Hello, after adding lights, the sensors cannot properly scan objects after packaging , What went wrong and do you have any good suggestions。
[image]

_Originally posted by @stone778 in https://github.com/Field-Robotics-Japan/UnitySensors/issues/110#issuecomment-1926182263_

Autumn60, 2024-02-07:
@stone778
Please describe the conditions under which the bug occurs or publish a branch that reproduces the environment.

**Example of conditions**
* Environment: [Editor or Build] on [Windows or Linux]
* Unity Version: 2021.3
* Unity Project: [UnitySensors or others]
* UnitySensors Version: [git tag or branch name]

stone778, 2024-05-08:
Environment: Windows (Project environment) + Linux (ROS environment)
Unity Version: 2021.3.32f1c1
Unity Project: UnitySensors
UnitySensors Version: UnitySensors-2.0.0b

I have already tried the solutions provided by other contributors on GitHub, such as setting shaders, adding lights, configuring soft shadows, etc. However, even after packaging, the sensors are still not functioning. Please help me identify the issue.
[image]
[image]
[image]
[image]

RyodoTanaka, 2025-03-04:
I represented the same error with following commit.
- Unity Version: 2023.3.49f1
- UnitySensors Version: master branch

The reason of this issue comes from the mistake of shader programming.
And that problem will be fixed with #188 PR.

[image]
[image]