Field-Robotics-Japan/UnitySensors #115: How to simulate the MMwave radar sensor?

chu12345, 2024-02-07:
Which script is used to simulate the MMwave radar sensor?

Autumn60, 2024-02-09:
For data such as point clouds, LiDAR sensor implementations will be helpful for millimetre wave radar implementations.
For the functionality to obtain the speed of the target object, implementation is expected to be difficult.
This is because _GetComponent_ cannot be used in the _RaycastCommand_ function.

RyodoTanaka, 2024-02-13:
@chu12345

I agree with @Autumn60 regarding the multi-echo feature of the MMwave sensor.
Multi-echo as defined here refers to the ability to also measure the distance of objects hidden behind the object from the sensor's illuminated position. As @Autumn60 explains, I don't have an idea to implement this feature in the current Unity functionality either.

However, I think it is possible to implement it if the information is only about the most intense objects, without considering multi-echoes. 
I have seen many examples of so-called Radar sensor or Range sensor implemented in other simulators.
I think @chu12345 is right that you are talking about sensors that are not multi-echo sensors.
If so, we would like to incorporate the implementation of those sensors in our implementation plan.
I await @chu12345 opinion !