Field-Robotics-Japan/UnitySensors #116: Direction of gravity vector of IMU simulation

yosuke, 2024-02-09:
When I visualize the IMU simulation using ROS RViz IMU plugin, the linear acceleration vector is directing downward (toward the ground), while this should direct upward (toward the sky) as shown in the following URL:
http://wiki.ros.org/rviz_imu_plugin?action=AttachFile&do=get&target=rviz_imu_plugin.png

This is probably because the gravity vector is ***added*** in the UnitySensors implementation:
https://github.com/Field-Robotics-Japan/UnitySensors/blob/master/Assets/UnitySensors/Runtime/Scripts/Sensors/IMU/IMUSensor.cs#L51

While the gravity vector is ***subtracted*** in the other implemenation (e.g. gazebo):
https://github.com/gazebosim/gazebo-classic/blob/gazebo11/gazebo/sensors/ImuSensor.cc#L375-L376

We probably need to think "Because the IMU is not moving, linear velocity is applied ***against*** the gravitational force (that is why the vector should be subtracted)".

Autumn60, 2024-02-09:
@yosuke 
Thank you for the Issue.

This is due to my lack of knowledge about the IMU.
For reference, what values of vertical acceleration does the IMU show when it is in free fall?

yosuke, 2024-02-09:
@Autumn60
Good point.
My assumption is, it turns zero during free fall.
And yes, it seems right:
https://physics.stackexchange.com/questions/402611/why-an-accelerometer-shows-zero-force-while-in-free-fall

Autumn60, 2024-02-10:
@yosuke 
This issue was fixed in #118.
Would you try the develop branch?

RyodoTanaka, 2024-02-13:
@yosuke @Autumn60 
Thank you for your discussion and the merge.
This issue looks already solved by the merge, thus I close it.

PR #118 (2024-02-10): Fix/imu grav dir
* Fix #116
* Modified to cache normalised gravity vectors.
* Due to problems caused by UnityEngine physics calculation intervals, each value was changed to be calculated in FixedUpdate().
closed 2024-02-13 by RyodoTanaka