Field-Robotics-Japan/UnitySensors #121: GNSS仕様変更の提案

Autumn60, 2024-02-11:
English translations will be added later.

緯度経度座標の計算元である直交座標を、
**ワールド座標系でのGNSSSensorの座標**・**GNSSSensorがワールド座標系原点の緯度経度高度の設定を持つ**
から
**特定のオブジェクトのローカル座標系でのGNSSSensorの座標**・**特定のオブジェクトがそのオブジェクトの緯度経度高度の設定を持つ**
に変更する．

つまり、現行が「ワールド座標系原点の緯度経度高度を設定する・緯度経度の方向はXZ軸で固定」であるのに対し「任意の位置の緯度経度高度を設定できる・緯度経度の方向は変更可能」にする。
+α 複数のGNSSSensorで共通の座標系を持たせることが容易になる。

RyodoTanaka, 2024-02-13:
# English Translation

I'll get it sorted out as it is a blurb.

The Cartesian coordinates from which the latitude and longitude coordinates are calculated,
GNSSSensor coordinates in the world coordinate system, GNSSSensor has the latitude, longitude, and altitude settings of the world coordinate system origin.
From.
GNSSSensor coordinates in the local coordinate system of a specific object, with the specific object having the latitude, longitude, and altitude settings of that object.
The GNSSSensor coordinates in the local coordinate system of a specific object have the latitude, longitude and altitude settings of that object.

In other words, "the latitude and longitude altitude of the world coordinate system origin is set, and the latitude and longitude direction is fixed in the XZ axis" whereas the current "the latitude and longitude altitude of any location can be set, and the latitude and longitude direction can be changed".

It will be easier to match the coordinate system to the model of the loaded city or to have a common coordinate system for multiple GNSSSensors.

Autumn60, 2024-02-18:
#138 にて実装

closed 2024-02-18 by Autumn60