Field-Robotics-Japan/UnitySensors #157: The bug about the definition of ROS message in this package, and the suggestion for the fix

TaroABEJido, 2024-04-30:
I found a few bugs when I tried to use Unity Sensor on ros 2.
The Unity outputs the error below when executing the sample program (Demo_imu_ros and Demo GNSS ros etc..)

```
Assets\UnitySensorsROS\Runtime\Scripts\Serializers\Tf2Msgs\TFMessageMsgSerializer.cs(39,50): error CS1061: 'HeaderMsg' does not contain a definition for 'seq' and no accessible extension method 'seq' accepting a first argument of type 'HeaderMsg' could be found (are you missing a using directive or an assembly reference?)
```

Then, I found the cause of the issue in [this source code](https://github.com/TaroABEJido/UnitySensors/blob/master/Assets/UnitySensorsROS/Runtime/Scripts/Serializers/TF/TFMsgSerializer.cs "TFMessageMsgSerializer.cs") and fixed it.

I want to suggest a fix by Pull Req. 
I hope that this fix is helpful for this project.

Autumn60, 2024-05-02:
I have confirmed that the proposed PR will resolve the issue.
Thank you for raising the Issue and PR!

closed 2024-05-02 by Autumn60

sawan-kcl, 2024-05-15:
Hi, 
Sorry to continue on a closed discussion. I'm getting this error when i change my protocol from ROS1 to ROS2. Please let me know how can i solve this?

PR #158 (2024-05-02): Fix MsgSerializer
fix #157