Field-Robotics-Japan/UnitySensors #162: IMU linear acceleration is relative to origin and not sensor

Jarhead20, 2024-07-01:
I added an IMU sensor and put it on a 45 degree angle, then the game object was dragged in the editor to generate some data. The below image shows the data captured in RQT and shows both the X and Y linear accelerations changing. The expected behaviour is to have only the X or the Y linear accelerations change as it would be on a physical IMU. The current behaviour shows the acceleration relative to the origin and not the sensor itself. I am using version 2.0.4.

[image]

Jarhead20, 2024-07-01:
https://github.com/Field-Robotics-Japan/UnitySensors/assets/62244227/3dbf8a3a-148e-47f8-9384-52137f280dc1

Jarhead20, 2024-07-01:
Another related problem is that since the IMU rotation is relative to the orign, when the imu starts, it isn't initialised to 0 degrees. Instead it starts at whatever the angle is in the simulator. I would suggest a button to toggle between relative and absolute rotation.

Autumn60, 2024-07-01:
@Jarhead20 
Thanks for the Issue report!! 🙏 

You are right.
IMU angles and angular velocities do not seem to be relative.

As I am busy this week, I will fix this next week.

Jarhead20, 2024-07-04:
I forked the repo and have relative motion mostly working, I will finish my SLAM implementation to properly test it and I have also made a wheel encoder sensor. I also added the option to remove gravity and switch between absolute and relative modes. I'll create a pr for each once I have fully tested them.

Autumn60, 2024-07-06:
@Jarhead20 

The implementation of IMUSensor.cs has been reviewed.
I am ashamed to say that I have little knowledge of IMU sensors and it is difficult to judge whether this is working properly.

May I ask you to check that the behaviour is what you expect?

https://github.com/Autumn60/UnitySensors/tree/fix/imu_relative_calc

Jarhead20, 2024-07-10:
I will be able to check the functionality of the sensor this time next week. In the meantime, you could compare the functionality of your code to what I was testing with. What I wrote seems to act like a real IMU sensor so if your code produces the same output then it should be good. It's a bit messy because it has my work on the encoder in the same branch. https://github.com/Field-Robotics-Japan/UnitySensors/compare/master...Jarhead20:UnitySensors:master