Field-Robotics-Japan/UnitySensors #165: publishing RAW images

sawan-kcl, 2024-07-11:
Hello,
The work with ROS sensors has been going well so far but i have encountered a bottleneck. In order to use the depth images and RGB images for 3D mapping, i need data in RAW format instead of compressed format.
Is there any way that the camera and depth sensors publish images in raw format? currently it publishes in compressed format with JPEG encoding.

Also, I plan to use IMU but in zero gravity/microgravity environment in Unity. Could that be a problem?

RyodoTanaka, 2024-07-18:
You can decompose compressed image to raw image using `image_transport` package.
https://index.ros.org/p/image_transport/

About IMU in zero gravity environment, I have not tried that.
Thus, I do not know what the problem will be.

sawan-kcl, 2024-07-18:
thanks @RyodoTanaka for your response,
I have been using `image_transport` till now but this comes with processing overhead resulting in lower frame rates of raw images. The maximum frame rate i get from setting `frequency` in ros sensor prefabs is 45 Hz which and after passing through `image_transport` raw image frequency is 25 Hz.
I wanted higher frame rates that's why i was wondering if there's any way to directly get raw images.

RyodoTanaka, 2024-10-09:
@sawan-kcl 
Recently, we merged the raw image publish function with https://github.com/Field-Robotics-Japan/UnitySensors/pull/171 .
Thus, please try to use `beta/v2.x.x` branch, and you do not need to use `image_trnsport` package.