Field-Robotics-Japan/UnitySensors #168: error when using CustomMsg

niuwenju, 2024-08-26:
SysCommand.publish - Unknown message class 'Livox/CustomMsg'
UnityEngine.Debug:LogError (object)
Unity.Robotics.ROSTCPConnector.ROSConnection:ReceiveSysCommand (string,string) (at C:/Users/niuwj.DIGITALCHINA/Desktop/ros_unity/ROS-TCP-Connector/com.unity.robotics.ros-tcp-connector/Runtime/TcpConnector/ROSConnection.cs:680)
Unity.Robotics.ROSTCPConnector.ROSConnection:Update () (at C:/Users/niuwj.DIGITALCHINA/Desktop/ros_unity/ROS-TCP-Connector/com.unity.robotics.ros-tcp-connector/Runtime/TcpConnector/ROSConnection.cs:587)

Not registered to publish topic '/livox/lidar'! Valid publish topics are: dict_keys([]) 
UnityEngine.Debug:LogError (object)
Unity.Robotics.ROSTCPConnector.ROSConnection:ReceiveSysCommand (string,string) (at C:/Users/niuwj.DIGITALCHINA/Desktop/ros_unity/ROS-TCP-Connector/com.unity.robotics.ros-tcp-connector/Runtime/TcpConnector/ROSConnection.cs:680)
Unity.Robotics.ROSTCPConnector.ROSConnection:Update () (at C:/Users/niuwj.DIGITALCHINA/Desktop/ros_unity/ROS-TCP-Connector/com.unity.robotics.ros-tcp-connector/Runtime/TcpConnector/ROSConnection.cs:587)

Autumn60, 2024-10-07:
Sorry for late reply.

A little more information is needed to find out about errors.
Could you please tell me what steps I need to take to reproduce the error?

Xin-Lee, 2025-02-07:
@Autumn60 Hi, I have the same error.

I first merged 5 commits in feature/livoxCustomMsgPublisher branch to V2.0.5, and then I tried to use RaycastLiDARLivoxCustomMsgPublisher.cs to replace the RaycastLiDARPointCloud2MsgPublisher.cs on the Lidar object.

When I play the scene, I get these two errors in Unity Editor:

[image]
```
SysCommand.publish - Unknown message class 'Livox/CustomMsg'
UnityEngine.Debug:LogError (object)
Unity.Robotics.ROSTCPConnector.ROSConnection:ReceiveSysCommand (string,string) (at ./Library/PackageCache/com.unity.robotics.ros-tcp-connector@c27f00c6cf/Runtime/TcpConnector/ROSConnection.cs:680)
Unity.Robotics.ROSTCPConnector.ROSConnection:Update () (at ./Library/PackageCache/com.unity.robotics.ros-tcp-connector@c27f00c6cf/Runtime/TcpConnector/ROSConnection.cs:587)
```

and these errors in ros_tcp_endpoint:
```
[ERROR] [1738901316.750438]: Failed to resolve message name: No module named 'Livox'

[ERROR] [1738901317.238861]: Not registered to publish topic '/livox/lidar'! Valid publish topics are: dict_keys([]) 
```

I don't know whether the customMsg publisher can be used now? If not, is there any way I can implement this function?

Xin-Lee, 2025-02-12:
> [@Autumn60](https://github.com/Autumn60) Hi, I have the same error.
>
> I first merged 5 commits in feature/livoxCustomMsgPublisher branch to V2.0.5, and then I tried to use [...]

I got the answer, this error is because I didn't add CustomMsg and CustomPoint into ROS workspace. First I add them by following [this tutorial](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/setup.md#install-unity-robotics-demo) and then change to proper k_RosMessageName in CustomPointMsg.cs andCustomMsgMsg.cs