Field-Robotics-Japan/UnitySensors #191: A guidance seems necessary and useful

HaiCLi, 2025-03-10:
Hi,

After I install this unity package in engine, I didn't know how to use it. When I try to pull the "demo" into scene. The following content is shown.

[image]

Thanks for your help!

RyodoTanaka, 2025-03-11:
@HaiCLi 
Thank you for your issue.
I have not met this error.
How did you get our UnitySensors package, from PackageManager ?

HaiCLi, 2025-11-04:
> [@HaiCLi](https://github.com/HaiCLi) Thank you for your issue. I have not met this error. How did you get our UnitySensors package, from PackageManager ?

I found the problem. remove the "#3.0.0" when import