Field-Robotics-Japan/UnitySensors #195: How to correctly use the Depth Camera Sensor?

ZPHuang-ACM, 2025-04-08:
To test the `Depth Camera Sensor`, I created a simple test environment with just one cube in the scene and added a `Depth Camera Point Cloud2 Msg Publisher` script to the `DepthCamera` prefab came with the package. However, it seems my point clouds are not transformed correctly and there are a lot of noisy points.  Are there any additional steps I need to follow to use the sensor? I am using Unity 6000.0.33f1/6000.0.23.f1 and `UnitySensors v2.0.5`

[image] 
Simple test environment

[image]
Visualizing  pointclouds in MeshLab

[image]
Visualizing pointclouds in Rviz2

RyodoTanaka, 2025-04-10:
@ZPHuang-ACM 
Thank you for opening this issue.

Honestly, when we use depth camera (depth buffer inside of Unity), we still have problem that we could not get correct depth in 3D.
That causes this kind of problem.

I do not have any solution immediately.
We should keep on to find this.

If you have any idea or insights about this, even if its not the solution, we are very welcome to hear it.

Bests,

Bob-Eric, 2025-04-10:
But I tested `UnitySensors v2.05` under Unity version `2023.2.20f1c1` and it doesn't seem to be a problem!

[image]

RyodoTanaka, 2025-04-10:
@Bob-Eric 
Thank you for your report !

This means, the problem is occurring on Unity6 version.

@ZPHuang-ACM 
If possible, I would like to ask you to use 2023.3.x version for brief fix.

ZPHuang-ACM, 2025-04-16:
> But I tested `UnitySensors v2.05` under Unity version `2023.2.20f1c1` and it doesn't seem to be a problem!
> 
> [image]

ZPHuang-ACM, 2025-04-16:
> [@Bob-Eric](https://github.com/Bob-Eric) Thank you for your report !
> 
> This means, the problem is occurring on Unity6 version.
> 
> [@ZPHuang-ACM](https://github.com/ZPHuang-ACM) If possible, I would like to ask you to use 2023.2.x version for brief fix.

I am still very noisy point clouds. Do I need to

Bob-Eric, 2025-04-17:
> > [@Bob-Eric](https://github.com/Bob-Eric) Thank you for your report !
> > This means, the problem is occurring on Unity6 version.
> > [@ZPHuang-ACM](https://github.com/ZPHuang-ACM) If possible, I would like to ask you to use 2023.2.x version for brief fix.
> 
> I am still getting very noisy point clouds with 2023.2.x. I think I am missing some steps. Do I need to do [Migrating To URP/HDRP](https://github.com/Field-Robotics-Japan/UnitySensors/blob/master/MigrateToURP%26HDRP.md)? I didn't do this.

I think you could adjust the Gaussian Noise Sigma parameter in the DepthCameraSensor script.

closed 2025-04-22 by ZPHuang-ACM